adafruit/Adafruit_CircuitPython_SD #2: MemoryError importing adafruit_sdcard

johnbeaulieu, 2017-09-24:
I am running an M0 Adalogger:

Adafruit CircuitPython 2.0.0-3-g3ad01dd on 2017-09-20; Adafruit Feather M0 Adalogger with samd21g18
>>> import adafruit_sdcard
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError:

sometimes I get:
MemoryError: memory allocation failed, allocating 199 bytes

I have the adafruit_bus_device directory and the adafruit_sdcard.py file in the lib folder on the root of the device.

Also repeatable on CircuitPython 2.0.0 release version

tannewt, 2017-09-25:
Hi! I know its tight on memory. Could you try the mpy files from the bundle available [here](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/latest)? mpy files take less memory to load.

johnbeaulieu, 2017-09-25:
That did it. Thank you.

tannewt, 2017-09-25:
Great!

closed 2017-09-25 by tannewt