adafruit/Adafruit_CircuitPython_SD #3: unable to mount SD Card on ESP8266

jerryneedell, 2017-10-09:
I have an adalogger featherwing with the SDCARD CS set to in GPIO2 and I am unable to mount the card using this driver. I have had the same results as below with CS on GPIO15 and with a 2.4" TFT with an ADCARD - (It uses GPIO2- so I am using it on the adalogger to simplify switcheing between test setups). I also have the same isssue using an SDCard Breakout. 
Here is the code I use to attempt the mount:

As noted below, I am able to mount the card using a modified version of the micropython sdcard driver.

I have had no problems using the adafruit_sdcard driver on M0 boards.

```
import adafruit_sdcard
import busio
import digitalio
import board
import storage

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(board.GPIO2)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
```
results in:
```
Adafruit CircuitPython 2.0.0-8-gf498167 on 2017-10-08; ESP module with ESP8266
>>> import sdmount_adafruit
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdmount_adafruit.py", line 10, in <module>
  File "libraries/drivers/sdcard/adafruit_sdcard.py", line 84, in __init__
  File "libraries/drivers/sdcard/adafruit_sdcard.py", line 111, in _init_card
OSError: no SD card
>>> 
```

I copied the micropython sdcard.py  from https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py 
I  modified line 122 time.sleep(.050) instead of time.sleep_ms(50)

the following code successfully mounts the card on the ESP8266:
```import sdcard,machine
import busio
import digitalio
import board
import storage

# Connect to the card and mount the filesystem.
sd = sdcard.SDCard(machine.SPI(1), machine.Pin(2))
vfs = storage.VfsFat(sd)
storage.mount(vfs, "/sd")
```

jerryneedell, 2017-10-09:
I will be happy to follow up on this to see if I can figure out what the problem is. One obvious difference is in the baud rate settings in the 2 drivers for the card initialization.

jerryneedell, 2017-10-15:
I just wanted to provide an update on this.
I am not having much success in determining the problem other than it appears to be in the iteration between this driver and the HSPI driver on the ESP8266
When I look at the signals on a logic analyzer, I can see the expected operation for 2 cases:
ESP8266 SD Card mount using micropython scard.py
M0 SD Card Mount using adafruit_sdcard.py

however if I try to Mount the card on the ESP8266 using the adafruit_sdcard.py 
the signal responses are not normal and the mount fails. The outgoing command stream on MOSI is not the same as either case above. It appears to get shifted in time as it retries.
IN response, the SD Card never properly appears to enter SPI mode and there is no response at all on MISO.

I can get some response on MISO by setting wait=False when calling self._cmd in line 108, but the resulting MISO responses don't make a lot of sense and it still fails.

I have been trying to unravel the code for the ESP8266 HSPI interface and compare it to the M0 version, but it has not been easy to make sense of it yet. 

Does anyone with more experience with these drivers know if we should even expect this to work as written. That is, is adafruit_sdcard.py code only compatible with the M0 spinner driver and not with the ESP8266 driver?

tannewt, 2017-10-16:
This is fixed in the 2.x branch: https://github.com/adafruit/circuitpython/commit/6c7d35c40cb2b498b34befd4dcfc3d6d0bf3208b

It was a result of write_value not being the write value but rather a number of extra bits to clock. It was 255 which led it to confusing the 8 bit boundaries for commands.

closed 2017-10-16 by tannewt