adafruit/Adafruit_CircuitPython_SD #9: Trouble using Adafruit SD card library on Raspberry Pi 3 / Python 3.5

MyDev45, 2018-07-31:
Hello !

I'm currently trying to implement a Feather Adalogger board on a Raspberry Pi 3 B+. RTC part works fine as I'm able to set my RPi's clock from the adalogger board, but I'm having some troubles concerning SD functionnality.

I'm using the advices provided by Adafruit  Adalogger tutorial using CircuitPython, but not with the ".mpy" libraries because I could not get it recognized by Python (I'm using Python 3.5, and maybe ".mpy" librairies need micropython to be installed). I'm using the same libraries, but not precompiled (".py" files).

Here is the example code I'm trying to execute. It's the same as your example, but I'm having some troubles with the instruction `sdcard = adafruit_sdcard.SDCard(spi, cs)`. 

```
import os
 
import adafruit_sdcard
import board
import busio
import digitalio
import storage
 
# Use any pin that is not taken by SPI
SD_CS = board.D4
 
# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
```

The error says "readinto() got an unexpected keyword argument 'write_value'. This instruction is written in the "adafruit_sdcard.py" file at line 215, which is the source file from the Adafruit_CircuitPython_SD library (https://github.com/adafruit/Adafruit_CircuitPython_SD.

The complete instruction is `spi.readinto(self._single_byte, write_value=0xff)`

While trying to debug the problem, I thought the called method was the one contained into the "busio.py" file, but it seems that it's not this one, because putting comments on this method has no effect on my code, whereas I was expecting that Python would not find the method. I also tried to debug with pdb, but I can't step into the target file of this "readinto()" instruction.

Have you got any idea about how to solve this problem?

Thanks a lot,

R.L.

jerryneedell, 2018-07-31:
The SPI module for "Blinka" is still a "work in progress" - I think this issues is related to one filed previously https://github.com/adafruit/Adafruit_Blinka/issues/17 and the ultimate resolution will depend on how the SPI module it is be completed.

MyDev45, 2018-07-31:
Hello jerryneedell,

Yes I'm using "Blinka" to get it working, otherwise it's not possible at first glance. Looking at the Blinka SPI code, it does not redefine such a readinto method, and so I did not really understand why putting comments on this method in the "busio.py" Blinka file didn't generated an error... Looks like my error is effectively quite close to yours in the other issue.

thanks you for answering :)

ladyada, 2018-07-31:
i think you'd be better off just storing to the raspi SD filesystem instead of another SD card, no? im not sure we do or will support external SD card on the Blinka layer

MyDev45, 2018-08-01:
Hello ladyada !

Actually I'm interested in storing other files on a second SD card. I can do it using USB key, but SD card could be interesting for my application. As RPi3 has SPI pins, I thought it was possible to manage a device using these pins, and why not another SD card. For RPi3, this is just a SPI slave in theory, but maybe the library is not suited to such an application on RPi3. I've investigated a lot before posting the message, but the error is quite strange, cause I can't find the method that is called by this line.

Thanks for answering !

ladyada, 2018-08-01:
yeah its not suited and linux has a lot of filesystem stuff it does, the best best thing you can do is use a USB sd reader/writer like this:  https://www.adafruit.com/product/939

tannewt, 2018-08-02:
@MyDev45 Try changing https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/raspi_23/spi.py#L51

MyDev45, 2018-08-17:
Hi @tannewt ,

Thanks for your answer, and sorry for the delay. I've been trying to change this file, and also others, because the method is defined in many files, and it's difficult to trace during the execution. But unfortunately it's not the solution, because another "readinto" method with other arguments "start" and "end" is used some lines later, highlighting other errors. Maybe it will be an improvement in the next commits of this library. 

Thanks again

ladyada, 2018-08-17:
we've added start/end to blinka, make sure you have the latest version :)

kattni, 2018-12-03:
Hello. As there has been no activity on this in months, I'm going to close it. Please feel free to continue this thread if you have more questions. We can reopen it if necessary!

closed 2018-12-03 by kattni