adafruit/Adafruit_CircuitPython_SD #11: _cmd() hangs waiting for 0xfe after large write

dhalbert, 2018-10-05:
This forum thread provoked this issue: https://forums.adafruit.com/viewtopic.php?f=60&t=141707. The poster had this problem with all the SD cards they tried and I had trouble with the one I tried as well. The failures are not consistently at the same place across different cards or even multiple trials, so it appears to be timing-related (probably due to something the SD card is doing).

After the second or so `writeblocks()` of a 7168-byte buffer, a `readblocks()` follows, which calls 
`_cmd(12, wait=False)`. This hangs in `_cmd()` in this loop; it never receives a `0xfe`.

```
                            # Wait for the start block byte
                            while buf[1] != 0xfe:
                                spi.readinto(buf, start=1, end=2, write_value=0xff)
```
I tried setting `wait=True` and it made no difference.

Attached is an instrumented version of `adafruit_sdcard.py`, which produces the output below. Notice that a `0x7` is received first, which doesn't show up anywhere else, followed by `0xff` indefinitely. Also attached is the test program (test.py).
[adafruit_sdcard.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_SD/files/2448814/adafruit_sdcard.py.txt)
[test.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_SD/files/2448815/test.py.txt)

Output:
```
>>> import test

waiting for 0xfe 0x0 
readblocks 512
waiting for 0xfe 0x0 0xff 
readblocks 512
waiting for 0xfe 0x0 0xff 
readblocks 512
waiting for 0xfe 0x0 0xff 
readblocks 512
waiting for 0xfe 0x0 0xff 
readblocks 512
waiting for 0xfe 0x0 0xff 
WRITE 10
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
WRITE 10
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
WRITE 4106
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 3584
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
WRITE 4106
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 3072
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 512
writeblocks 512
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
readblocks 512
waiting for 0xfe 0x0 0xff 
WRITE 8202
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 7168
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 512
writeblocks 512
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 512
readblocks 512
waiting for 0xfe 0x0 0xff 
readblocks 512
waiting for 0xfe 0x0 0xff 
WRITE 8202
readblocks 512
waiting for 0xfe 0x0 0xff 
writeblocks 512
writeblocks 7168
readblocks 512
waiting for 0xfe 0x0 0x7 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff [[[typed ctrl-c here]]] Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 20, in <module>
  File "test.py", line 20, in <module>
OSError: [Errno 5] Input/output error
>>> 
```

0x1d00ffff, 2018-12-24:
Hello, I am seeing the same issue on the ItsyBitsy M4 Express. 

I captured a few data points using the SDcards I have and the results are different but repeatable for each card type:

#### Lexmark 4gb class 10 card: 
This one is interesting. Writing up to 16384 bytes at a time to a file is OK, but **only if it is the first write to the file**. Sample code:

This completes OK every time (truncate file and write):
```py
with open('/sd/test.bin', 'w') as output:
    bytes_out =output.write(bytearray([1] * 16384))
    print('WROTE', bytes_out)
```
This fails every time (write to a file already containing 10 bytes):
```py
with open('/sd/test.bin', 'w') as output:
    output.write(bytearray([1] * 10))
with open('/sd/test.bin', 'a') as output:
    bytes_out =output.write(bytearray([1] * 16384))
    print('WROTE', bytes_out)

```

#### Unbranded 8gb class 10 card:
Fails on a 1024 byte write, infinite 1023-byte writes are fine. Interestingly, it appears that it is the **second** 1024 byte write that fails, the first is successful.

Unbranded 8gb log:
```
...
WROTE 1022
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
WROTE 1022
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
WROTE 1022
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
waiting for 0xfe: 0x0 0xff 
WROTE 1024
waiting for 0xfe: 0x0 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff [[[ctrl+c here]]] Traceback (most recent call last):
```

devoh747, 2019-08-05:
I just tested this on my GrandCentral using my Pullrequest #19 (Samsung EVO 32GB fix) ..

...
WRITE 32778
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
WRITE 32778
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
WRITE 36874
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
WRITE 36874
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
WRITE 40970
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
WRITE 40970
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
waiting for 0xfe: 0x0 0xff 0xff 
WRITE 45066
Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "code.py", line 25, in <module>
MemoryError: memory allocation failed, allocating 45066 bytes

I did three runs and each time it died in the same spot "MemoryError: memory allocation failed, allocating 45066 bytes"

dhalbert, 2019-11-18:
Should be fixed by #20. Others have reported success.

closed 2019-11-18 by dhalbert