adafruit/Adafruit_CircuitPython_SD #13: Samsung EVO Select 32GB does not initialize with CircuitPython 3.1.2 on Feather Express M4 with Adalogger

Helitronix, 2019-03-20:
First reported here with all the details: https://forums.adafruit.com/viewtopic.php?f=60&t=147212

Basic summary is that the Feather Express M4 / Adalogger hardware works with other micro SD cards that I have now tried, but the Samsung EVO 32GB card fails to initialize with CircuitPython 3.1.2 and the February 4'2019 libraries.  The card is properly initialized if I use the SD Cardinfo sample application in Arduino mode.  So the hardware is working fine, but something isn't quite right in the CircuitPython stack with this card.

danhalbert asked that I submit the bug here.  Thanks for your help!

kattni, 2019-04-11:
Same issue reported on 4.0beta6 on PyPortal. #15

devoh747, 2019-07-24:
The issue still exists in CP 4.0.2 and CP 4.1.0.rc1 using the 20190723 4.x libraries.  I used a GrandCentral and a Samsung 32GB EVO.  I verified the same setup works under Arduino example code  "Datalogger"  that is for any SD card.

devoh747, 2019-08-04:
I have part of it figured out..

Some SD cards require the CRC info for CMD55 and ACMD41.. so in adafruit_sdcard.py
changing lines..

194  to     self._cmd(55, 0 , 0x65)
195  to     if self._cmd(41, 0x40000000, 0x77) == 0:

I'm not sure why line 195 was a _block_cmd before.. but making it a _cmd gets it past this. Also not sure why the value 0x200000 was used. The examples on the net use 0x40000000.

It works for my SanDisk 16GB like it did before...  There is still an issue but now it is a hang 
at line 156    if self._cmd(9, response_buf=csd) != 0:
for Samsung EVO 32GB..  I have not figured out why it hangs and that _cmd function is pretty difficult for a beginner to figure out.  I do see a bunch of differences in how buf is setup in _cmd and _block_cmd..    I hope this is helpful.

devoh747, 2019-08-05:
The issue is that Samsung EVO 32GB cards require all command codes to have a valid CRC. I used https://github.com/hazelnusse/crc7/blob/master/crc7.cc to calculate all CRC codes and it is now working. On line 195 I also changed one command from a _block_cmd to a _cmd. It would hang otherwise. This works for my SanDisk 16 GB card which worked before the fix as well and also works on my Samsung EVO 32GB. Please test this with all cards. The spec does say that by default all command code CRC values are ignored except CM0, CMD8 and I believe ACMD41.. It's just that Samsung decided to ignore the default and check all command CRC values.. I was not even able to force it off. Please test the heck out of this. My first fix!

dhalbert, 2019-11-18:
We believe #20 fixes this.

closed 2019-11-18 by dhalbert

Helitronix, 2019-11-20:
@devoh747 thanks for jumping in to fix the issue.  I appreciate your effort to do this.