adafruit/Adafruit_CircuitPython_SD #15: unable to access Samsung EVO Plus microSDHC UHS-I card 32G from PyPortal with CircuitPy and adafruit_sdcard library

ddigioia2, 2019-04-11:
Any help is appreciated, and please let me know if you need any more details.
Unfortunately, this is currently the only microSD card I have.
I also reformatted the card (full, slow format) with the official app from https://www.sdcard.org/downloads/formatter/

Thanks.
David

Using mu-editor serial console:

Adafruit CircuitPython 4.0.0-beta.6 on 2019-03-30; Adafruit PyPortal with samd51j20
import board
import busio
import digitalio
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = digitalio.DigitalInOut(board.SD_CS)
import adafruit_sdcard
import storage
sdcard = adafruit_sdcard.SDCard(spi, cs)

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_sdcard.py", line 113, in __init__
  File "adafruit_sdcard.py", line 147, in _init_card
  File "adafruit_sdcard.py", line 203, in _init_card_v2
OSError: timeout waiting for v2 card

kattni, 2019-04-11:
This is a duplicate of #13. This is a known issue. Please feel free to add more information to issue #13, however I am closing this issue in favor of the original.

dhalbert, 2019-04-11:
This was a followup I requested from this thread: https://forums.adafruit.com/viewtopic.php?f=60&t=149712

kattni, 2019-04-11:
Understood. Both issues now reference each other, that is sufficient.

dhalbert, 2019-11-18:
We believe #20 fixes this.

closed 2019-11-18 by dhalbert