adafruit/Adafruit_CircuitPython_SD #21: "no SD card" error

JHDillenb, 2019-08-20:
Hello,

I have been attempting to use this repo in order to read an exterior micro sd card through a Jetson Nano. The wiring is set up to use SPI1. And I am following the example code:

"""
import adafruit_sdcard
import busio
import digitalio
import board
import storage
# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(board.SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
# Use the filesystem as normal.
with open("/sd/test.txt", "w") as f:
    f.write("Hello world\n")
"""

However, I have ran into several issues that I would like to ask for help on. 
1. 'import storage' comes back as not a module. Is this a file living in the repo somewhere?
2. adafruit_sdcard.SDCard(spi, cs) will return with an error on line 113 (self.__init_card())
3. As well as another error on line 141, which is the line raising the "no SD card" error, but I am suspecting that it could be a syntax error? I haven't seen an if statement in python formatted that way before, but of course I am willing to be corrected on that and anything thing else you see.

JHDillenb

ladyada, 2019-08-20:
this library isn't supported for linux dev boards. and the nano SPI isnt well tested, please use the Nano's native SD card or a USB SD reader!

JHDillenb, 2019-08-21:
I see. That is unfortunate for me, but thank you very much for informing me on this, ladyada.
You wouldn't happen to know of any other libraries that I may want to check on as a substitute for this one, would you?

ladyada, 2019-08-21:
i dont know of any user-space SD card libraries. its...just not something that linux was designed to do.

dhalbert, 2019-11-18:
Closing as not applicable.

closed 2019-11-18 by dhalbert