adafruit/Adafruit_CircuitPython_SD #27: OverflowError: long int not supported in this build

michelkluger, 2019-11-13:
When I import this to the Adafruit CircuitPython 5.0.0-alpha.5 on 2019-11-04; Adafruit Feather M0 Adalogger with samd21g18 I get the mentioned error,

OverflowError: long int not supported in this build

Do  you know how I can fix that? Or what might be causing it?

dhalbert, 2019-11-13:
We have longints turned off on the non-Express builds for space reasons. Could you give me the rest of the error message (line numbers, etc.)? I'll see if there's a possible workaround. This may or may not be related to https://github.com/adafruit/circuitpython/issues/2267

michelkluger, 2019-11-13:
when I have only the line:

import adafruit_sdcard

 in my main.py, I will get this error already:

OverflowError: long int not supported in this build

dhalbert, 2019-11-13:
The basic problem is a large constant `0x40000000` in the code. Will see if we can work around this.

PR #29 (2019-11-18): Avoid a longint as a cmd arg
This fixes #27.

PR #26 introduced a longint as a cmd arg, which makes the library fail on non-longint builds, such as Feather M0 Adalogger. I reworked the `_cmd()` method to take either an integer or a `bytes`, so non-longint constants can be expressed in the code.

@michelkluger, would you like to test? I can supply a .mpy if you need one.
closed 2019-11-18 by dhalbert