adafruit/Adafruit_CircuitPython_SD #38: Example not working in 6.1.0-beta.2 (OK in 6.0.0)

anecdata, 2020-12-05:
Adafruit PyPortal with samd51j20
6.1.0-beta.2 on 2020-12-03
adafruit-circuitpython-bundle-6.x-mpy-20201202

The SD part of my code has been stable for ages, but updating from CircuitPython 6.0.0 to 6.1.0-beta.2 triggers this error (trace shown here after running the library simpletest example):
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 20, in <module>
  File "adafruit_sdcard.py", line 116, in __init__
  File "adafruit_sdcard.py", line 137, in _init_card
  File "adafruit_sdcard.py", line 125, in _clock_card
AttributeError: 'SPIDevice' object has no attribute 'spi'

Press any key to enter the REPL. Use CTRL-D to reload.
```

anecdata, 2020-12-05:
Oh, I guess `sdcardio` is the way to go now. Easy swap. Is this library deprecated, or are there advantages to using it over `sdcardio`?

dhalbert, 2020-12-05:
This is failing because CircuitPython 6.1.0-beta.2 includes https://github.com/adafruit/circuitpython/pull/3612, which moves `adafruit_bus_device` to a native module. That module does not provide properties to replicate `SPIDevice.spi` and similar attributes, which are set by the Python version. We should think about whether there is a lot of code that depended on those properties, and whether that code is "wrong" and should be fixed.

anecdata, 2020-12-05:
Looks like SD is the only bundle library with this particular construction, not sure what other patterns would break drop-in replacement of `adafruit_bus_device` library with native:
```
$ egrep -r 'spi\.spi' adafruit-circuitpython-bundle-py-20201205
adafruit-circuitpython-bundle-py-20201205/lib/adafruit_sdcard.py:        while not self._spi.spi.try_lock():
adafruit-circuitpython-bundle-py-20201205/lib/adafruit_sdcard.py:        self._spi.spi.configure(baudrate=self._spi.baudrate)
adafruit-circuitpython-bundle-py-20201205/lib/adafruit_sdcard.py:            self._spi.spi.write(self._single_byte)
adafruit-circuitpython-bundle-py-20201205/lib/adafruit_sdcard.py:        self._spi.spi.unlock()
adafruit-circuitpython-bundle-py-20201205/lib/adafruit_sdcard.py:            self._spi.spi, self._spi.chip_select, baudrate=baudrate, extra_clocks=8
```

jepler, 2021-01-27:
We believe this was fixed in the core.  Please feel free to re-open if it affects 6.1.0 or 6.2-alpha.

closed 2021-01-27 by jepler