adafruit/Adafruit_CircuitPython_SD #39: Working with FT232H and spi-sd breakout board?

Gerd1986, 2021-01-06:
Hi guys,

i have the adafruit ft232h breakout board and one of the 5 dollars amazon sd via spi breakout boards. it is not even doing the initialization and failing with "no sd card".
Did you test the sd library with the ft232h breakout board?

Kind regards,

Gerd

jepler, 2021-01-27:
We've also reproduced this with the [Adafruit FT232H](https://www.adafruit.com/product/2264) breakout & [Adafruit MicroSD card breakout](https://www.adafruit.com/product/254).  At this time, I don't have a resolution to the problem.  @caternuson has also reproduced the problem.

In my case I tested on a Debian Linux x86_64 system. What I observed is that the serial input data is always 0xff.

Some notes from my investigation:

* It is necessary to use a GPIO for the CS pin, rather than pin D3 ("CS0") which is an automatically-managed select line.  This is important because the SD card initialization process must send out 0xff bytes with the chip select line HIGH.
* SPI clocking looks right (250kHz during initialization phase); SPI data out looks right.
  * I did not look at clock phase/polarity, but I don't think SD is special here
* Voltages all look appropriate
* I also tried adding a pull-up resistor on MISO as suggested in some internet sources (doesn't seem like the problem, since data comes in as 0xff without it, but I don't know how FT232H pull ups work), it didn't make a difference.
* I tried adding extra delays during initialization, in case a PC is faster than CircuitPython and that leads to a problem.
* I looped MOSI back to MISO and verified that I received the same as I sent, so it's not a general problem with SPI input on the FT232H.

makermelissa, 2021-01-27:
Maybe try some various brands of SD cards. I've had different results across brands before.

jepler, 2021-01-28:
We've identified that it's an incomplete implementation of busio.SPI in Adafruit_Blinka for the FT232H chip.  Other Blinka implementations of busio.SPI may have the same limitation.  You can follow the above-linked bug for more information.

jepler, 2021-02-08:
This should be fixed by a blinka update

closed 2021-02-08 by jepler

Gerd1986, 2021-02-12:
Its working now! Great work!