adafruit/Adafruit_CircuitPython_SD #41: Possible conflict between SD module and MPU6050 module

gtseidler, 2021-01-25:
I'm trying to do datalogging for the MPU6050 accelerometer sensor.  Specifically I'm trying to stream data to SDcard on a MKRzero running the latest circuit python.

It looks like I'm running into a conflict between the mpu6050 module (or something I2C?) and the sdcard module.

I'm able to run the sample code for the MPU6050 and get correct acceleration data back via serial.  I'm also able to run usual sample code that simply writes a txt file to the microSD card on the MKRzero.

However, when I include all needed modules in the otherwise working example for the 6050 (see at bottom of this post), so that its imports are now:

import time
import board
import busio
import adafruit_mpu6050
import adafruit_sdcard
import storage

then I get memory errors when it is saved to to the MKRzero.  When mpu6050 is imported first I get a memory error in sdcard at line line 494.  Reversing the order of imports, so that adafruit_sdcard is imported before adafruit_mpu6050 results in a memory allocation error instead at line 55 of the mpu_6050 module.

Any ideas?  With thanks.  For completeness, below is the MPU6050 code that works as shown, but then fails when the sdcard module is included in the imports.

`# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import board
import busio
import adafruit_mpu6050

i2c = busio.I2C(board.SCL, board.SDA)
mpu = adafruit_mpu6050.MPU6050(i2c)

while True:
    print("Acceleration: X:%.2f, Y: %.2f, Z: %.2f m/s^2" % (mpu.acceleration))
    print("Gyro X:%.2f, Y: %.2f, Z: %.2f degrees/s" % (mpu.gyro))
    print("Temperature: %.2f C" % mpu.temperature)
    print("")
    time.sleep(1)`

ladyada, 2021-01-25:
there's not enough memory for your entire project, you can use Arduino or upgrade to a chip with more SRAM, the m0 has only 32KB you likely need 64kB

closed 2021-01-25 by ladyada

gtseidler, 2021-01-25:
Thanks, that makes sense.  A follow-up question: is there anywhere that I can find the SRAM needs for each of the circuit python modules?  That'll help me figure out in advance what projects can, or cannot, be done with the limited memory on the MKRzero (or other boards).

ladyada, 2021-01-25:
add the filesizes of the mpy's and maybe add 20? there no specific way to know