adafruit/Adafruit_CircuitPython_SD #47: storage library

HourGlss, 2022-04-21:
attempting to use on raspberry pi using https://learn.adafruit.com/circuitpython-on-raspberrypi-linux?gclid=EAIaIQobChMIr8zA54ak9wIVQ__jBx2CfgkJEAAYASAAEgKCyPD_BwE

ran python3 -m pip install adafruit-circuitpython-sd

and then tried to use example but cannot find storage? please advise.

dhalbert, 2022-04-21:
This is for use on an SPI bus connected to an SD card. On a Raspberry Pi, you can simply use an SD card to USB adapter, and use regular file-based I/O that is a part of Python (`open()`, etc.)

dhalbert, 2022-04-21:
For more support, ask on https://adafru.it/discord or https://forums.adafruit.com. GitHub is for bug or enhancement requests.

closed 2022-04-21 by dhalbert

HourGlss, 2022-04-21:
No. I am not attempting to use the SD CARD. I want to use the storage module. I understand that there is an SD CARD. OBVIOUSLY there is an sd card. https://docs.circuitpython.org/en/latest/shared-bindings/index.html that indicates that storage is available on raspberry pi 4 which im using.

it is not. Please reopen this issue as I believe it to be a bug. please do not simply close my issue without addressing it.

dhalbert, 2022-04-21:
`storage` is not available via Blinka. Please ask for further support in discord or the forums. It's not a bug in this library.

dhalbert, 2022-04-21:
Your edited comment makes things clearer, but it's still the case that `storage` is not available on Blinka. Equivalent functionality is available using an SD card adapter, so you don't need to wire up SPI to talk to an SD card.

HourGlss, 2022-04-21:
I dont want to use an sdcard adapter it has to work via spi.

dhalbert, 2022-04-21:
Let's discuss this in discord please.

HourGlss, 2022-04-21:
any idea on where i can find the github issues page for the https://docs.circuitpython.org/en/latest/shared-bindings/support_matrix.html page?

dhalbert, 2022-04-21:
resolved in discord