adafruit/Adafruit_CircuitPython_SD #52: Library uses short argument names

tekktrik, 2022-11-08:
`pylint` suggests using argument names with at least 3 letters.  This library uses argument names of shorter length, and while these warnings have been disabled for now, they should be considered for renaming.  This may require the rework of Learn Guides and other references to code snippets.

tekktrik, 2023-03-28:
@dhalbert @jposada202020 @kattni This was closed because `.pylintrc` was updated, but I believe the intention for this issue (as well as the others like it with the same name) is to consider refactoring at a later date.  Just wanted to start a discussion as to whether this should be reopened (and if so, should the previous PR maybe be reverted since it just disables it at that `.pylintrc` level).

dhalbert, 2023-03-28:
@tekktrik For this particular library, the short name was `cs`. I believe this name is fine, and that `pylint`'s three-character minimum rule is not so sensible in this case. `CS` is the silk labeling on most boards, and we try to match that. Similar names would be `tx` and `rx` in `busio.UART`.

jposada202020, 2023-03-28:
Yes. I will update the other ones as well, as you could see also in the BME280 PR, the discussion was held, and `cs` exception was a better choice for this particular case. PR #55 was done with chip_select, but I reverted after discussing it with danh in discord. Cheers.

tekktrik, 2023-03-28:
I mostly bring it up because a possible refactoring was something @kattni and I had previously discussed but didn't know if/when it might happen, so it may be worth resolving whether that's still on the table before those issues are closed - they were there to mark those libraries for it when they were caught during a pylint update a while ago.

kattni, 2023-03-28:
@tekktrik This issue appears to be specific to the short variable names and Pylint taking issue with them. The refactor issues were worded differently, if I recall correctly. 

@jposada202020 Was `cs` the only Pylint short-name issue in this library?

If yes, then I believe this issue should remain closed. There should be a separate issue regarding a possible future refactor. If there is not, @tekktrik please see to it that it gets added. 

Thanks all!

jposada202020, 2023-03-28:
@kattni Indeed only the cs was the case, I took a look, to similar issues in other libraries and as far as I can tell I think they are related with the `cs` value.